AngeloCasi/FUGU-ARDUINO-MPPT-FIRMWARE #4: Battery voltage discovery and excessive overcharge voltage /Sate of charge reading

tytower, 2021-10-04:
The problem with most mppt chargers is that when they read the battery voltage they don't turn off the background charging current. This results in an elevated voltage reading on the line and messes the calculations around. 
Somehow you must first turn the charge off then read the battery voltage ,then turn the charge back on. 

Secondly with Lithium , the charge voltage must never be greater than the number of cells in the battery times 3.65 volts . This results in damage to the cells if the plate voltage gets too much "push force" EMF. 16 cells for instance ,the top push must not go over 16 x 3.65v =58.4 volts .
If it does it has been shown by scientific experiment that 10 % over causes lithium dendrites and 20% over even causes the copper plate to start building copper dendrites too , both of which pierce the separator and short the cell.

Thirdly the SOC of a Lithium battery ,especially LiFePo4 ,LFP , cannot be done by a simple division . The charging graph at say 3.3 volts is almost a straight line . Only the coulomb counting method will have any chance of being close.
[image]

X-Ryl669, 2021-11-04:
You shouldn't use this code to charge a Lithium battery, not even a Lead battery since there isn't any CC mode implemented (only CV).

tytower, 2021-11-04:
If it is constant voltage it can be used on both . As long as the current being delivered is less than a figure of C/2 on the battery it will be fine because on CV the current drops quickly. I don't think you know what you are doing really.

tytower, 2021-11-04:
**All I am suggesting to you though is that the charge source needs to be turned off in your code before you take a voltage reading and then turned back on.**

JBecker32, 2022-04-17:
not necessary, because the battery voltage will only be **over**estimated with charge current flowing. This is not a problem because it will not harm the battery or the charging algorithm in any way, it will just make the output curve 'softer'. Battery voltage is not used in any way to calculate/estimate SOC, this has to be done by coulomb counting.

JBecker32, 2022-04-17:
> If it is constant voltage it can be used on both .

no, using CV in lead acid battery charging can be dangerous as lead acid batteries tend to show a thermal runaway on high charging currents/voltages (due to temperature increase) which are often used with solar power charging to use as much of the available solar power as possible. I would be very careful with this (due to bad experience!).

tytower, 2022-04-17:
> no, using CV in lead acid battery charging can be dangerous as lead acid batteries tend to show a thermal runaway on high charging currents/voltages (due to temperature increase) which are often [...]

Clearly you are not aware that you car alternator is a constant voltage charger. Nowadays it holds a voltage of 14 volts and the current is varied according to the state of the battery. So wake up , thats what most battery chargers do.

tytower, 2022-04-17:
> not necessary, because the battery voltage will only be **over**estimated with charge current flowing. This is not a problem because it will not harm the battery or the charging algorithm in any [...]

Well I disagree ,you should try to be accurate with what the voltage of the plates is at, to set the input mppt volyage .If you don't do that then you are just guessing are'nt you. 
If you are charging LFP batteries voltage is a very big factor and coulomb counting while good must be discarded once the voltage rises over 3.4 volts . You definitely want to stop charging when the voltage reaches 3.5 volts because if you get up to 3.65 volts you start destroying the SEI layer by clogging it with Ions.

JBecker32, 2022-04-17:
> > not necessary, because the battery voltage will only be **over**estimated with charge current flowing. This is not a problem because it will not harm the battery or the charging algorithm in any way, it will just make the output curve 'softer'. Battery voltage is not used in any way to calculate/estimate SOC, this has to be done by coulomb counting.
> 
> Well I disagree ,you should try to be accurate with what the voltage of the plates is at, to set the input mppt voltage .If you don't do that then you are just guessing are'nt you. If your MPPT voltage exceeds 3.65 volts you will destroy the cells over time. If you are charging LFP batteries voltage is a very big factor and coulomb counting while good must be discarded once the voltage rises over 3.4 volts . You definitely want to stop charging when the voltage reaches 3.5 volts because if you get up to 3.65 volts you start destroying the SEI layer by clogging it with Ions.
> 
> 3.4 volts or at worst 3.5 volts are clear voltage indicators of a full battery. The same applies to Li Ion cells whose charging should be stopped at 4.1 volts for the same reasons.

???
The MPPT voltage is the voltage of the solar generator,this has nothing to do with the charge voltage of the battery.
And yes, cell voltage is a very important factor in charging Lithium batteries (as is for most other battery chemistries). But if you set the (measured) charge voltage to a certain limit (let us say 3.42V/cell for LiFePo4), this is the maximum voltage the battery will reach. The difference between charger output voltage and cell voltage is just given by the resistance of the cabling and cell connectors multiplied with the charge current. As charge current decreases when the cell voltage reaches the set output voltage the cell will not be overcharged.
There is no reason why 'coulomb counting must be discarded once over 3.4V'. The coulomb efficiency of Lithium cells  is nearly 100% and is not depending on cell voltage (below the absolute maximum allowed cell voltage ranges).
All this becomes a little bit irrelevant by the fact that lithium batteries are **always** used with a BMS which will cut off the battery from the charger in any case of cell overvoltage (temperature, current, etc) and which will also do the coulomb counting itself.

JBecker32, 2022-04-17:
> > > If it is constant voltage it can be used on both .
> > 
> > 
> > no, using CV in lead acid battery charging can be dangerous as lead acid batteries tend to show a thermal runaway on high charging currents/voltages (due to temperature increase) which are often used with solar power charging to use as much of the available solar power as possible. I would be very careful with this (due to bad experience!).
> 
> Clearly you are not aware that you car alternator is a constant voltage charger. Nowadays it holds a voltage of 14 volts and the current is varied according to the state of the battery. So wake up , thats what most battery chargers do. Some have a battery destroying feature called "equalisation" where the voltage is upped to 18 volts or more for a few minutes and that helps destroy them quickly so they sell more .

JBecker32, 2022-04-17:
> If it is constant voltage it can be used on both .
>
>
> no, using CV in lead acid battery charging can be dangerous as lead acid batteries tend to show a thermal runaway on high charging [...]

Sorry, but you clearly have no idea about professional charging of an (often deep cycled!) lead acid energy storage battery.

hasan33-1492, 2023-12-12:
> @tytower or others I see comments about the 1C variations and temperature variations. Are there proposed changes to the code? I understand a lead acid battery wants a 3 stage charge, as well I understand that MPPT is dumping as much energy as possible into the storage batteries, and I think this thread is talking about changing from MPPT once you reach the 3rd stage voltage, and you switch to a trickle charge instead of MPPT max energy.
> 
> I've copied the design making it a Kicad which I intend to have JLCPCB assemble. I'm mostly interested in lead acid, as you can recouped the lead and reuse the lead over time. I'm also likely to use other chemistry instead of sulfuric acid.
> 
> I see that there is an unused ADC on the initial design, should I include a battery temperature sensor in my copy of this board?
> 
> My copy can be found here https://github.com/jharvey/Charge_Controller are there any suggested changes? I added a couple minor things, like reverse polarity protection on the MPPT input and the battery output. As well I added some USB surge suppression. Are there any other items I should include in my copy of this board?

Nice!