AngeloCasi/FUGU-ARDUINO-MPPT-FIRMWARE #14: Add schematics and PCB layout to this repository

renne, 2021-12-06:
Please add schematics and PCB layout to this repository.

swat2go, 2024-10-29:
This is exactly what i miss as what can a piece of software archieve without something in media res

cheezemanrich, 2024-10-29:
It's in the Instructables website.
https://www.instructables.com/DIY-1kW-MPPT-Solar-Charge-Controller/

swat2go, 2024-10-30:
Thanks a lot 

> It's in the Instructables website. https://www.instructables.com/DIY-1kW-MPPT-Solar-Charge-Controller/

read.me **should** be the place or, even better, wrap it all together in one place e.g. here, with a hint to the original place instead littering around in so many places.  If just one goes down .... webarchive might help but ...

The instruct article is huge, partially in pretty detail, but horrendous sorted and structured to find anything of value between rectum output what a MPPT, sync / async  aso is for. Thats a job for wiki. I wouldn't look at it if not already knowing. 

This repo, jose pineiro nor the article answer the question if it does buck-boost or simple cheap buck-down. 

After delving into the article reading on backflow / body diode i assume that this piece might be brilliant, but not satisfying my requirements as it shuts off when Usolar =< Ubatt but which is the designed default to me. 

I need / work on a MPPT buck-**BOOST** for a 2 or 4 Panel array, each 600W design from 12V - 60V in, up to 48V/+... out to reduce the cabling and mounting costs, work hassle with thick cables & connectors. single panel monitoring / failure, temp, ... are welcome benefits. Inverters also benefit from higher DC voltages as current heats them up, +++++ 

E. g. a 2k4W array of 4 panels each 600W at 40V (Vmpp is usually even lower) delivers 60Amps boah 16mm² (eurelectric) or thicker cable is recommended 8-(  compared at 200V only 12Amps remain, satisfied by cheap 1,5mm², for equal power. That is hell of a lot easier to handle, connect, cheaper ++++ but including light eyes on contact. So some extra safety above 60V has to be taken into account for fire fighters, service, ....
This is especially if the array is in a distant from poi, battery or inverter like roof, garden ....

All done in unlimiting interworking open source      KiCAD  SPICE   FreeCAD    to give anyone in the world a chance to participate regardless from OS, (expensive) commercial software, .... and benefit from renewable energy.

cheezemanrich, 2024-10-30:
This project hasn't been updated in over 3 years.
Stop complaining about free information. You didn't spend the time to create it. If you don't like it, create it yourself.

Here is someone who spun off this project, see issue #30:
https://github.com/jharvey/Charge_Controller

You're welcome!