AngeloCasi/FUGU-ARDUINO-MPPT-FIRMWARE #17: P&O Algorithm

Shaopus, 2021-12-13:
Hi
I have a problem with mppt algorithm about your code, as follow：
```c
if(powerInput>powerInputPrev && voltageInput>voltageInputPrev)     {PWM--;}  //  ↑P ↑V ; →MPP  //D--
else if(powerInput>powerInputPrev && voltageInput<voltageInputPrev){PWM++;}  //  ↑P ↓V ; MPP←  //D++
else if(powerInput<powerInputPrev && voltageInput>voltageInputPrev){PWM++;}  //  ↓P ↑V ; MPP→  //D++
else if(powerInput<powerInputPrev && voltageInput<voltageInputPrev){PWM--;}  //  ↓P ↓V ; ←MPP  //D--
else if(voltageOutput<voltageBatteryMax)                           {PWM++;}  //  MP MV ; MPP Reached - 
```
About P&O Algorithm, `if(powerInput>powerInputPrev && voltageInput>voltageInputPrev) `, should increase PWM.
However，your code is the opposite of it, Can you explain?
Thank you！

Tomsan83, 2021-12-27:
i think it is ok, when power goes up and voltage as well, you are climbing on the curve of power on the right drection  so you need to decrease PWM. You increase PWM when moving left and decrease when moving right. It is because the fact when voltage is MAX, the ciuurnet is MIN and opposite. For detail go to Angelo project page: 
https://www.instructables.com/DIY-1kW-MPPT-Solar-Charge-Controller/
chart called PV iV graph op.regions.

noeldum, 2025-03-30:
I have the exact same request as the OP. Thank you for your comment @Tomsan83 but it still does not make sense to me.

Assuming the polarity of PWM is such that PWM++ increases the output voltage then:

When both power and voltage increase then we are on the left side of the curve and we have just moved to the right. Hence MPP is located further to the right at higher voltage. Hence I would code  PWM++ here instead of

`if(powerInput>powerInputPrev && voltageInput>voltageInputPrev)     {PWM--;} `

Can someone explain this. What are we missing here?

[image]

noeldum, 2025-03-30:
Sorry. Just answering myself here.

My assumption on the polarity of PWM was wrong. PWM++ decreases the output voltage. It is actually well indicated in green on the chart. (D++ moving to the left).

It's all good and understood now.