AngeloCasi/FUGU-ARDUINO-MPPT-FIRMWARE #21: GPIO init typo?

tditwister, 2022-03-27:
https://github.com/AngeloCasi/FUGU-ARDUINO-MPPT-FIRMWARE/blob/fd4d73f1aea06d6b2ea0354d22c86279f60c432e/ARDUINO_MPPT_FIRMWARE_V1.1.ino#L315

I'm new to this but shouldn't this be TempSensor instead of TS?

jaybee79, 2022-04-22:
I would agree with you, TS is intended to contain the temperature (as float value). The correct pin numer is set there: 

> #define TempSensor      35          //SYSTEM PARAMETER - Temperature Sensor GPIO Pin

As GPIOs on microcontrollers are normally acting as input input at boot-time/after reset and the GPIO 0 pin is only used as input in this firmare there is nothing that could be damaged.
Anyway: that's a minor failure that should be corrected.

cyberangel22k, 2023-07-22:
`  if(sampleStoreTS<=avgCountTS){                               //TEMPERATURE SENSOR - Lite Averaging
    TS = TS + analogRead(TempSensor);
    sampleStoreTS++;   
  }
  else{
    TS = TS/sampleStoreTS;
    TSlog = log(ntcResistance*(4095.00/TS-1.00));
    temperature = (1.0/(1.009249522e-03+2.378405444e-04*TSlog+2.019202697e-07*TSlog*TSlog*TSlog))-273.15;
    sampleStoreTS = 0;
    TS = 0;
  }`
I think the sketch is correct.