AngeloCasi/FUGU-ARDUINO-MPPT-FIRMWARE #27: ideas for alternate backflow circuit

jbaltes, 2023-03-24:
I wonder about the complicated backflow control circuit - after read your notes at instructables I could understand your decision a bit.
Did you test a charge pump solution (PWM from microcontroller, totem-pole driver)?

Here are some ideas how the problem could be solved easier - did you test anything about this...
* use PMOS channel mosfet, higher Rdson - not preferred because of higher power losses
* use hotswap controller as gate driver - they are not as expensive as the isolated DCDC-converter, I think
* change the backflow NMOS from high side switching to low side switching - why don't you switch the negative input, it would be so much easier. The problem that then arises with the input voltage measurement can be solved with a simple operational amplifier.

fl4p, 2023-07-12:
Hi,
Using PMOS in power application appears to be uncommon, due to high Rds(on) as you mentioned.

Most commercial MPPT chargers I've seen use common ground (NMOS + charge pump). Common ground would not allow low side switching. Although, I don't see any strong reason for a common ground in a solar-battery application. Not having a common ground would also allow low side current sense, which is much simpler to implement when using shunt resistors.

I'd put the backflow control on the battery side. This way we can cut off the battery, if something is wrong. In case of a short high-side mosfet of the buck converter, we can use the low side mosfet to prevent excess voltage output to the battery by just shorting the solar input (half-bridge shoot through). Without the backflow switch on the battery side this would burn the battery fuse.

fl4p, 2023-07-30:
Update:
The most simple /cheap drive circuit is probably a diode from the HS Gate to the Backflow Gate. This way the backflow switch turns on as long as the buck converter is running.