AngeloCasi/FUGU-ARDUINO-MPPT-FIRMWARE #30: Permission to copy the schematics and pcb design in KiCAD and publish it.

Istria1704, 2023-10-02:
Hello Angelo,

Thank you very much for sharing this awesome project with the world. I learned a whole lot about synchronous buck converters and mppt, all thanks to your instructable.

I'm wondering if you would find it ok if I would copy your design in KiCAD and put it on Github (with full credits to you of course). I noticed there is no free version of the Proteus PCB Design software. And I would like to mod your design a bit for higher output current (50-60A).

Planning to build a solar boat with 2S5P solar panel configuration which would give a Voc of 100V and a Imp of 50A. The output would be a 48V lead-acid battery bank and a 3 kW electric motor. So while cruising, I'd need around 40A of output current for the motor plus some charging current if the battery is not full. 

I intend to design (copy) the pcb in KICAD, and if it's ok with you, I'll share the tested design on github once it's tested and proven reliable for a while.

Thanks again for sharing this project!

P0rkR0ll, 2024-01-10:
Hello @Istria1704 , I was just wondering if you ever started work on this. I would like to contribute as well and build off this ref design!

Istria1704, 2024-01-10:
No, but not long after I opened this issue, I came across someone who had already done exactly that (copied it to KiCAD). I'll see if I can find it again later. If I do I'll post the url. It might even be one of the issues here.